JelmerT/cc2538-bsl #186: Cannot get Sonoff dongle into bootloader mode

greew, 2024-12-31:
Hi all.

I've had the "SONOFF ZigBee 3.0 USB Dongle Plus (ZBDongle-P)" for a few years now. Never having flashed a new firmware onto it, it still has the original 20210708 firmware, when when I received it.

I now have a new Zigbee devices, that requires a newer firmware, hence I downloaded the newest firmware from the Koenkk repo and want to flash it. 

BUT - I simply cannot get the Dongle into bootloader mode. 

I have read the Readme here and also other threads around of how to use this script, and I simply cannot see what I'm doing wrong.

```
[cc2538-bsl] python3 cc2538-bsl.py -ewv ~/Downloads/CC1352P2_CC2652P_launchpad_coordinator_20240710.hex -V --bootloader-sonoff-usb
sonoff
Opening port /dev/ttyUSB0, baud 500000
Reading data from /home/js/Downloads/CC1352P2_CC2652P_launchpad_coordinator_20240710.hex
Firmware file: Intel Hex
Connecting to target...
*** sending synch sequence
Traceback (most recent call last):
  File "/home/js/code/cc2538-bsl/cc2538-bsl.py", line 1162, in <module>
    if not cmd.sendSynch():
  File "/home/js/code/cc2538-bsl/cc2538-bsl.py", line 384, in sendSynch
    return self._wait_for_ack("Synch (0x55 0x55)", 2)
  File "/home/js/code/cc2538-bsl/cc2538-bsl.py", line 277, in _wait_for_ack
    raise CmdException("Timeout waiting for ACK/NACK after '%s'"
CmdException: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
```

I'm not used to play around with print boards and such, but I got "desperate" and went out and bought a PH00 screwdriver and wanted to try to enable the bootloader by unplugging the dongle, hold down the bootloader button, insert the dongle again and keep the button pressed for at least 10 seconds (one time I tried 30 seconds), but still I cannot get the script past the `Synch` command, where it crashed then. 

Now, does anyone have any other ideas of how to get the dongle into bootloader mode, or should I just consider the dongle "broken"? 

Thank you in advance - and a happy new year! :)

breuerfelix, 2025-04-07:
I got the same Problem. Any solutions found so far?

greew, 2025-04-07:
Nope.. I haven't found any solution..

tom-jm69, 2025-04-16:
Im facing the same Problem:

```bash
(env) ➜  cc2538_bsl git:(main) ✗ sudo chown tom:tom /dev/ttyUSB0 && python3 cc2538_bsl.py -p /dev/serial/by-id/usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_f4a00a8ae473ef119553e71e313510fd-if00-port0 -evw --bootloader-sonoff-usb CC1352P2_CC2652P_launchpad_coordinator_20240710.hex --bootloader-sonoff-usb
sonoff
Opening port /dev/serial/by-id/usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_f4a00a8ae473ef119553e71e313510fd-if00-port0, baud 500000
Reading data from CC1352P2_CC2652P_launchpad_coordinator_20240710.hex
Your firmware looks like an Intel Hex file
Connecting to target...
Traceback (most recent call last):
  File "/home/tom/Documents/github/cc2538-bsl/cc2538_bsl/cc2538_bsl.py", line 1161, in main_cli
    if not cmd.sendSynch():
           ~~~~~~~~~~~~~^^
  File "/home/tom/Documents/github/cc2538-bsl/cc2538_bsl/cc2538_bsl.py", line 383, in sendSynch
    return self._wait_for_ack("Synch (0x55 0x55)", 8)
           ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tom/Documents/github/cc2538-bsl/cc2538_bsl/cc2538_bsl.py", line 276, in _wait_for_ack
    raise CmdException("Timeout waiting for ACK/NACK after '%s'"
                       % (info,))
CmdException: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/tom/Documents/github/cc2538-bsl/cc2538_bsl/cc2538_bsl.py", line 1297, in <module>
    main_cli()
    ~~~~~~~~^^
  File "/home/tom/Documents/github/cc2538-bsl/cc2538_bsl/cc2538_bsl.py", line 1292, in main_cli
    if QUIET >= 10:
       ^^^^^
UnboundLocalError: cannot access local variable 'QUIET' where it is not associated with a value
```

The LED starts blinking green/red and then as soons as i start the script it stays red and the script crashes...

Djennez, 2025-05-15:
Not sure if the 2 errors in this thread are _exactly_ the same, but I found it through the `QUIET` error in the error from @tom-jm69 

Anyway, I got past that error by stopping all my Home assistant docker containers. Previously I had only stopped zigbee2mqtt. I have no idea if that was actually the cause (and the USB device was still passed to one of them) but it might be worth a try if you're in the same situation. Good luck!